ArminJo/micronucleus-firmware #1: your fork

cpldcpu, 2019-11-23:
Hi Armin,

it is true that the current maintainers of Micronucleus do not manage to dedicate the time to the project it deserves. If you want to become part of the maintainer team you are very welcome to!  (I don't think @bluebie @AHorneffer object). The main ask would be that some continuity is kept in revisioning and versions are tagged accordingly. I think creating a fork, especially with the message you added, will only add to confusion.

Best regards,
Tim

Bluebie, 2019-11-23:
Yes agreed. Would prefer new maintainers on the existing project instead of a fork

ArminJo, 2020-04-17:
Hi together,
**I am very sorry** but I just stumbled upon this issue I was not aware of.

I created a fork, since I got no answer to
[New ENTRYMODE ENTRY_POWER_ON #153](https://github.com/micronucleus/micronucleus/issues/153)

and no hex file for
[Disconnecting micronucleus leave USB -D pullup active #134](https://github.com/micronucleus/micronucleus/issues/134)

and no solution for 
[User program won't start without a pull-up #123](https://github.com/micronucleus/micronucleus/issues/123)

as well as no reaction to the MCUSR restrictions 
[MCUSR cleared - no information about reset source #111](https://github.com/micronucleus/micronucleus/issues/111)

And I could implement the integration of creating the upgrades directly by using make (I forgot the reference).
And what is about the ["new" digistump version]( https://github.com/ArminJo/micronucleus-firmware#installation-of-a-better-optimizing-compiler-configuration).

These are a lot of changes and I was quite pessimistic to believe to have any chance to put them to the mainstream especially with the experience I made so far.

So my question is: Will I have any chance to integrate all these, being a maintainer?

Best regards and again sorry for the late reaction
Armin

closed 2020-04-17 by ArminJo

cpldcpu, 2020-04-30:
@ArminJo I added you to the team now. I think you need to accept the invite. Looks like Github changed some things lately.