ArminJo/micronucleus-firmware #2: Unknown USB Device (Device Descriptor Request Failed) when running sketch without USB communication

psitech, 2020-04-15:
Hi Armin,
When programming "upgrade-t85_default.hex" [V2.5, commit f90621b] into a DigiSpark and then uploading the DigiSpark "Start" example, USBtreeview shows the DigiSpark Bootloader for 6 secs and then shows Unknown USB Device (Device Descriptor Request Failed):

[image]

I programmed the bootloader by: micronucleus.exe --run upgrade-t85_default.hex.
Micronucleus V1.11 showed only the Digispark Bootloader for 5 secs and then disappeared from USBtreeview [with sketch "Start" uploaded].
Is this a bug or is this intended behaviour with Micronucleus V2.5?

Regards,
Paul

psitech, 2020-04-15:
I subsequently programmed "upgrade-t85_entry_on_reset_no_pullup.hex" to see whether that binary made a difference. No, it shows the same Unknown USB Device.
However, now I'm NOT able anymore to re-program back to "upgrade-t85_default.hex"? The Digispark Bootloader is not visible in USBtreeview after insertion...
Did I brick my Digispark module?

Regards,
Paul

ArminJo, 2020-04-17:
Hi Paul,
I am very sorry, but I got the Error with upgrade-t85_entry_on_reset_no_pullup.hex last days too.
Now I found the bug in the upgrade program and fixed it.
The Digispark is only halfways bricked, only the bootloader is corrupt and you must burn it with avrdude.
Sorry again, and I will have a look at the Unknown USB Device problem next week.

Regards
Armin

ArminJo, 2020-04-17:
Hi Paul,
on my W10 I have no `Unknown USB Device (Device Descriptor Request Failed)` entry neither in the Device manager nor in Nirsoft's USBDeview.
What Digispark do you use and are there any extra componets attached to the USB pins?
Armin

psitech, 2020-04-17:
Hi Armin,
No worries, I was able to unbrick my Digispark using an AVRispMkII and AVRdude:

avrdude -c avrispmkII -P usb:0000B0005835 -p t85 -U flash:w:micronucleus-1.11.hex -U lfuse:w:0xe1:m -U hfuse:w:0x5d:m -U efuse:w:0xfe:m -v <enter>.

Digispark board is operating normally again.
I'm using a genuine Digispark micro-usb board without any extra components.
USBDview output:

[image]

Regards,
Paul

psitech, 2020-04-17:
Some more information that may help you:
- After unbricking the part by reflashing with "micronucleus-1.11.hex", pin 5 on the ATtiny85 is no longer the Reset pin but a pin you can use for other purposes. This is caused by the particular fuse setting above. This fuse setting is Digispark's default fuse setting. 
The side effect of this now-free pin 5 is that the AVRisp can't access the ATtiny85 anymore because the AVRisp requires the reset functionality to be present on the ATtiny85...
- After updating the bootloader with "upgrade-t85_default.hex" [V2.5, commit 0d7bef2] and no Arduino sketch loaded, the Digispark Bootloader is shown in USBtreeview and disconnects-connects every ~6 secs.
- When subsequently uploading the "Start" sketch, "Digispark Bootloader" is shown in USBtreeview for ~6 secs and then "Unknown USB Device (Device Descriptor Request Failed)" is shown permanently.

Regards,
Paul

ArminJo, 2020-04-19:
Hi Paul, 
wondeful, tha you have revived your Digispark.
Maybe I must check my laptops to reproduce the error.
Do you have an micro USB or a edge commector Digispark board, and do you se the error

psitech, 2020-04-19:
Hi Armin,
I have a genuine Digispark micro-USB board.
And yes, I still see the "Unknown USB Device (Device Descriptor Request Failed)" error when using your latest commit of "upgrade-t85_default.hex". See my "Some more information" paragraph above.
BTW, I'm on Windows 10 Pro.

ArminJo, 2020-04-19:
Hi,
I could reproduce it and updated the readme.
## Unknown USB Device (Device Descriptor Request Failed) entry in device manager
The bootloader finishes with an active disconnect from USB and after 300 ms setting back both D- and D+ line ports to input.
This in turn enables the pullup resistor indicating a low-speed device, which makes the host try to reconnect to the digispark.
If you have loaded a sketch without USB communication, the host can not find any valid USB device and reflects this in the device manager.
You can avoid this by actively disconnecting from the host by pulling the D- line (see below) to low.
E.g. using it as a tone() output pin and generating a short beep at startup will work. 

## Periodically disconnect - connect if no sketch loaded
This is a side effect of enabling the bootloader to timeout also when traffic from other USB devices is present on the bus.
It can be observed e.g. in the old 1.06 version too.

ArminJo, 2020-04-20:
Hi Paul,
disconnecting sets a pin to output active low and draws additional 2 to 3 mA through the pullup.
But it does not add any functionality and the bootloader cannot judge, if the sketch uses USB communication.
So this is something you definitely do not want in a bootloader!
Better pull it low by yourself, **now that it is documented**.

closed 2020-04-20 by ArminJo

psitech, 2020-04-20:
Hi Armin,
Fair enough. It's a choice. I just don't like unkown USB devices to show up in the device manager.
Perhaps I should have phrased the "disconnection from the host" slightly different. What I meant to say is that I would like to see the DigiSpark Bootloader disconnect from the host once a sketch is started. That sketch could then be a sketch with or without USB communication. A sketch with USB communication would then reconnect as the appropriate USB device, e.g. HID keyboard.
This is how I think it works with Micronucleaus V1.11.
Anyway, I'm fine with closing the issue.

Thanks,
Paul

ArminJo, 2020-04-20:
I agree, a unkown USB device is nasty, but an activated pin drawing current also.

The first is more failsafe and can be actively be avoided by the user program.
Maybe we need a new configuration STAY_ACTIVELY_DISCONNECTED.

History:
Micronucleus 1.06 just does no disconnect, 
Micronucleus 1.11 leave the D- active low / disconnected (which is no standard at all, but works for most USB adapters).
Micronucleus 2.x does a disconnect and then change D- pin to input, resulting in an unknown device.

psitech, 2020-04-21:
Hi Armin,
Just to let you know that:
- driving the D- pin low results in not showing the Unknown USB Device,
- I was not able to repeat the 6 secs disconnect/connect with the latest upgrade-t85_default.hex  [commit 7a48a8b ].

Regards,
Paul

ArminJo, 2020-04-21:
Hi Paul,
Thanks. It works as intended, so the software is deterministic :-)
Regards
Armin

ArminJo, 2020-05-11:
Hi Paul,
I optimized the bootloader again, and skipped the reconnect at exit.
So no disconnect/connect are happening and no unknown device is seen in the device manager, the host thinks of a libusb device even after the bootloader stops listening.

psitech, 2020-05-11:
Hi Armin,
Hmm, I still see the unknown device in USBtreeview when I upload this sketch [using the latest _entry_on_power_on_no_pullup_fast_exit_on_no_USB_ bootloader]:
```
void setup() {
  pinMode(1, OUTPUT);   // LED on Model A  or Pro
}

void loop() {
  digitalWrite(1, HIGH);
  delay(100);
  digitalWrite(1, LOW);
  delay(100);
}
```

ArminJo, 2020-05-11:
Hi Paul,
I checked it again at 2 W10 PC's and there is no unknown device.
Is the internal LED now blinking for you during the first 5 seconds?
This is an indication for the new version.

psitech, 2020-05-11:
Hi Armin,
No, the internal LED is not blinking during the first 5 seconds.
OK, I wil try to upload upgrade-t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex by means of the micronucleus commandline, and not by the Arduino IDE.

psitech, 2020-05-11:
Did this: micronucleus.exe --run upgrade-t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex .
The LED now flashes very quickly immediately after power-on but it stops flashing every second for a few 100 msecs. Then after ~6 secs the 100/100msec flash from my program above starts to show.
Then did a micronucleus.exe --erase-only and the LED is now showing the fast LED bursts indefinitely. Also USBtreeview shows the Digispark Bootloader continuously.
Next step I did: micronucleus.exe --run upgrade-t85_default.hex. And uploaded my user program. After uploading my user program starts immediately, but after unplugging & plugging the board, it takes the regular 6 secs to start the sketch. All the time USBtreeview shows the Digispark Bootloader.
The fast LED bursts are no longer seen.

psitech, 2020-05-11:
Also checked the fuses: Lfuse reads as 0xE1, Hfuse reads as 0x5D, Efuse reads as 0xFE. This is Digispark default.
Changed Hfuse to 0xDD [pin 5 now acts as reset pin] but that did not change anything. Still ~6 secs wait before my user program starts and USBtreeview shows the Digispark Bootloader continuously.

ArminJo, 2020-05-11:
Fantastic.
everything works as expected 😀 .
The t85_default has [no flashing LED](https://github.com/ArminJo/micronucleus-firmware#bootloader-memory-comparison-of-different-releases-for-t85_defaulthex) (this helps saving 64 bytes) and of course waits for 6 seconds.
Thanks a lot for feedback.

psitech, 2020-05-11:
Ah OK, so with upgrade-t85_entry_on_power_on_no_pullup_fast_exit_on_no_USB.hex , the fast LED bursts for the first~6 secs are intended?

ArminJo, 2020-05-11:
This bootloader is 64 bytes larger so there was place for the LED feedback.
See my table...

psitech, 2020-05-11:
OK, so "LED_MODE=ACTIVE_HIGH" means the LED burst? I did not understand that from the table :-) 
And "no_USB" means powered by external 5V, not connected to a USB host? I did read "no_USB" as a user program without USB communication...

Thanks anyway!

ArminJo, 2020-05-11:
OK,
you are totally right. 
I have to add a legend to this table!
Thanks!