ArminJo/micronucleus-firmware #5: Compile error

dantavares, 2020-07-01:
I'm having a problem compiling the firmware:

upgrade.c: Na função ‘write_new_bootloader’:
upgrade.c:94:23: erro: ‘bootloader_size’ undeclared (first use in this function)
   while ( page_addr < bootloader_size ) {
                       ^
upgrade.c:94:23: nota: each undeclared identifier is reported only once for each function it appears in
upgrade.c:98:32: erro: ‘bootloader’ undeclared (first use in this function)
       int subaddress = ( (int) bootloader ) + page_addr + word_addr;
                                ^
upgrade.c:99:33: erro: ‘bootloader_end’ undeclared (first use in this function)
       if ( subaddress < ( (int) bootloader_end ) ) { // valid boot code
                                 ^
upgrade.c:107:17: erro: ‘bootloader_address’ undeclared (first use in this function)
     erase_page( bootloader_address + page_addr );
                 ^
upgrade.c: Na função ‘forward_interrupt_vector_table’:
upgrade.c:153:39: erro: ‘bootloader_address’ undeclared (first use in this function)
     vector_table[ iter ] = 0xC000 + ( bootloader_address / 2 ) - 1;
                                       ^
make: *** [Makefile:117: upgrade.o] Error 1

dantavares, 2020-07-02:
Well, apparently only the upgrade part is failing to compile, so far I've solved this by copying the firmware to the upgrade folder for version 2.04 of the original maintainer "cpldcpu".

ArminJo, 2020-07-08:
There is an include file bootloader.h generated by the [makefile](https://github.com/ArminJo/micronucleus-firmware/blob/b917b67762dcb2d210587870d40c6ae098603fd3/firmware/Makefile#L151). Did you check its generation/existence?

closed 2020-07-16 by ArminJo