ArminJo/micronucleus-firmware #10: drop strings to save 28 bytes

nerdralph, 2020-08-23:
Although mn doesn't use vendor, product, or configuration strings, it still builds in string0 (language string descriptor).  Removing this code, saves 28 bytes.
You can test it without actually removing the code by changing the config:
```#define USB_CFG_DESCR_PROPS_STRINGS                 1```

drawkula, 2020-08-23:
Please let some space be there for making them unique on the bus (serial number?).

nerdralph, 2020-08-23:
> 
> 
> Please let some space be there for making them unique on the bus (serial number?).

What would be the point of that?  The upload tool only supports one device with the bootloader being plugged in.

drawkula, 2020-08-24:
...and the USB information for using e.g. a T85 as USB serial is different and hides elsewhere?

nerdralph, 2020-08-24:
> 
> 
> ...and the USB information for using e.g. a T85 as USB serial is different and hides elsewhere?

WTF are you talking about?  Micronucleus is a bootloader, not a USB-CDC implementation.

drawkula, 2020-08-24:
You're soooo friendly.
Thanks!

Ho-Ro, 2020-09-02:
@nerdralph Good proposal!

I've built a test version (default setup) with `#define USB_CFG_DESCR_PROPS_STRINGS 1` and its size drops under the page boundary -> 64 more bytes:
Available space for user applications: 6650 bytes

    Building Micronucleus configuration: t85_test
    Size of binary hexfile. Use the data size to calculate the bootloader address:
       text    data     bss     dec     hex filename
          0    1514       0    1514     5ea main.hex

    Building Micronucleus configuration: t85_default
    Size of binary hexfile. Use the data size to calculate the bootloader address:
       text    data     bss     dec     hex filename
          0    1542       0    1542     606 main.hex

commit 08d05a58 (2020-09-26): Saved 28 bytes by `#define USB_CFG_DESCR_PROPS_STRINGS 1`. Closes #10 re enabling push and pop of Y register in *usbdrvasm165.inc*. Closes #11.
commit 035381f3 (2020-09-26): Saved 28 bytes by `#define USB_CFG_DESCR_PROPS_STRINGS 1`. Closes #10 re enabling push and pop of Y register in *usbdrvasm165.inc*. Closes #11.
commit 27a01de6 (2020-09-26): Saved 28 bytes by `#define USB_CFG_DESCR_PROPS_STRINGS 1`. Closes #10 re enabling push and pop of Y register in *usbdrvasm165.inc*. Closes #11.
closed 2020-09-26 by ArminJo