ArminJo/micronucleus-firmware #11: USB_INT_VECTOR bug

nerdralph, 2020-08-24:
Y must be saved according to the avr-libc calling convention, but you commented out the push instructions to save it.
https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/usbdrv/usbdrvasm165.inc#L41

And if you comment out the in YL, SREG line, you should clear it to make sure you don't get false timeouts at line 55 based on what YL happens to be before the function is called.
https://github.com/ArminJo/micronucleus-firmware/blob/master/firmware/usbdrv/usbdrvasm165.inc#L55

nerdralph, 2020-08-24:
u-wire uses Z instead of Y, which makes sense since Z is a call-clobbered register.
https://github.com/cpldcpu/u-wire/blob/master/firmware/usbdrv/usbdrvasm.S#L23

nerdralph, 2020-08-24:
I've fixed it in my asm branch.  I decided to use X instead of Z, to leave Z free for possible future use by the asm main loop for lpm & spm.
Preliminary testing with t85_default and t85_aggressive seems OK.

closed 2020-09-26 by ArminJo

ArminJo, 2021-01-30:
Hi I just checked your usbdrvasm165.inc version, but it fails as well as replacing Y with X or Z, The only thing working for me was to comment out the Y push instructions, which may be against the rules but worked like a charm for me with default and aggressive.

nerdralph, 2021-03-08:
> 
> 
> Hi I just checked your usbdrvasm165.inc version, but it fails as well as replacing Y with X or Z, The only thing working for me was to comment out the Y push instructions, which may be against the rules but worked like a charm for me with default and aggressive.

I'm guessing you only copied the one file.  You'd need the whole thing given the other optimizations I did like using fixed registers for most globals.

ArminJo, 2021-03-11:
To clarify: I reactivated the push for Y!

commit 08d05a58 (2020-09-26): Saved 28 bytes by `#define USB_CFG_DESCR_PROPS_STRINGS 1`. Closes #10 re enabling push and pop of Y register in *usbdrvasm165.inc*. Closes #11.
commit 035381f3 (2020-09-26): Saved 28 bytes by `#define USB_CFG_DESCR_PROPS_STRINGS 1`. Closes #10 re enabling push and pop of Y register in *usbdrvasm165.inc*. Closes #11.
commit 27a01de6 (2020-09-26): Saved 28 bytes by `#define USB_CFG_DESCR_PROPS_STRINGS 1`. Closes #10 re enabling push and pop of Y register in *usbdrvasm165.inc*. Closes #11.