ArminJo/micronucleus-firmware #13: Avoid waiting for USB if D- is not pulled up

Ircama, 2020-10-11:
I think that if the USB D- pullup is removed to save energy, the USB should not work. At least, this is what I see from my tests.

With Digispark, in order to have the USB working when its socket is connected (device powered by USB) and at the same time no pull-up consumption without USB (e.g., with the device powered through +5V or VIN), the USB D- pull-up resistor should be directly connected to USB V+ and not to +5V: considering that there is a diode between USB V+ and +5V, the pullup will be powered (by USB V+) when the USB is connected and not applied when the USB is disconnected and the device is powered elsewhere (e.g., via Digispark 5V or VIN pins); this is because current does not flow from 5V to the pullup resistor but only through the opposite direction (which is from USB V+ to the pullup resistor directly and to +5V via the diode).

This means that if ENTRY_POWER_ON is selected (as well as START_WITHOUT_PULLUP) and the bootloader does not see PB3 pulled up at boot (which means that the USB is not connected and, in any case, there should not be the possibility to use USB without pullup), the bootloader should not try to start the USB (even within the current short time of the recommended configuration) and the LED should not flash at all.

Is there a way to set up such configuration as an upgrade to the [recommended configuration](https://github.com/ArminJo/micronucleus-firmware#recommended-configuration)?

ArminJo, 2020-10-22:
I documented some things [here](https://github.com/ArminJo/micronucleus-firmware#disconnecting-the-usb-d--pullup-resistor-marked-152-from-5-volt-vcc). 
If you have an idea how to reliably detect an extenal pullup at PB3, I can try to include it.

Ircama, 2020-10-22:
*fast_exit_on_no_USB* is good, but does not immediately start userprograms and it flashes the internal LED at boot; this might not be wanted if there is not a need to upload sketches (which follows the condition to disconnect the pullup resistor).

*fast_exit_on_no_USB_no_LED* is another possibility, but the LED is quite comfortable when uploading sketches; in this case, it is always disabled (also when the device is powered via USB and D- is pulled up through the diode).

I guess that adding `(USBIN & USBIDLE)` '*and*' condition to the [bootloader startup](https://github.com/ArminJo/micronucleus-firmware/blob/d5ef20176915e4bc85c08bdbbf4e351641a4a6e1/firmware/main.c#L358) can do the trick because PB3 should be configured as input in that place.

I can produce a PR in a short time if this concept is fine, adding a new flag that enables this new condition (e.g., *EXIT_IF_NO_PULLUP*).

At the moment I am not aware of configurations where the pullup resistor is missing and sketches can be anyway loaded, but I am not very sure of this. If there are no cases, this flag could replace *START_WITHOUT_PULLUP* because it totally disables the USB sketch uploading process where this flag is used (possibly some byte is also saved).

ArminJo, 2020-10-22:
Brilliant idea 🥇 
I named the entry condition ENTRY_D_MINUS_PULLUP_ACTIVATED and the configuration t85_entry_on_power_on_AND_USB_pullup_activated_fast_exit_on_no_USB
And disabling START_WITHOUT_PULLUP works too 😀
I pushed the version to the repo.
Please test it.

Ircama, 2020-10-22:
OK, it works. Let’s document the various use cases at power-on with this configuration:
- No user program installed (this condition occurs after installing or upgrading the bootloader), hardware pull-up configured on USB D-, USB connected or disconnected: the LED quickly flashes indefinitely (until a program is loaded). A user program can be loaded at any time. After the program is loaded, it immediately starts.
- No user program installed and no pull-up on USB D- (USB disconnected): LED off indefinitely. The user program cannot be loaded.
- User program installed with USB D- pulled up and USB connected: the LED flashes for about six seconds to wait for a new sketch to be loaded. After the timeout (or after loading a sketch), the user program starts.
- User program installed with USB D- pulled up and USB disconnected: the LED quickly flashes for about 600 milliseconds, then the user program starts.
- User program installed with no USB D- pull-up (USB disconnected): the user program immediately starts (without startup delay, without hang-up and without affecting the internal LED at boot).

In all cases, a hard reset (P5 grounded) immediately restarts the user program (the bootloader can only start at power-on).

ArminJo, 2020-10-22:
All right!

ArminJo, 2020-10-23:
I renamed configurations now.
You could do me a great favor, if you can make a short QA on the new layout.
Thanks
Armin

Ircama, 2020-10-26:
Hoping to help, I newly tested *upgrade-t85_entry_on_power_on_AND_USB_pullup_activated_fast_exit_on_no_USB.hex* (which works) and with a PR I provided some improvements to the documentation.

PR #16 (2020-10-26): Doc revision
Revised documentation of *ENTRY_D_MINUS_PULLUP_ACTIVATED_AND_ENTRY_POWER_ON*, *ENTRY_D_MINUS_PULLUP_ACTIVATED_AND_ENTRY_EXT_RESET*, recommended configuration and other topics.

close #13
closed 2020-10-26 by ArminJo