ArminJo/micronucleus-firmware #19: Error compilin a new configuration t85

freskpe, 2021-04-06:
Hello! I'm trying to compile a new configuration (ENTRY_JUMPER) on attiny85 but i got the next error... 
[image]

I'm using
Ubuntu 20.04
GNU Make 4.2.1
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

Thank you in advance!

TuxPuf, 2021-04-06:
I am getting the same error when using ENTRY_JUMPER but if I select other configurations it all compiles just fine. I'm unfortunately lacking the knowledge to help. I'm compiling on Windows 10 20H2 with Arduino 1.8.13 and the latest version of the board package from @ArminJo.

TuxPuf, 2021-04-06:
After some more "analysis" I found that the line 181 in bootloaderconfig.h seems to cause the compile error.
`  #define bootLoaderInit()   {JUMPER_DDR &= ~_BV(JUMPER_PIN); JUMPER_PORT |= _BV(JUMPER_PIN); _delay_ms(1);}`
this line is supposed to set the pullup on the jumper pin and add a delay to let the input stabilize so the pin isnt floating.

freskpe, 2021-04-06:
i'm was trying to compile another options (like ENTRY_ALWAYS) and i'm getting the same error, i think that the problem could be the compiler or another thing that i no know.

TuxPuf, 2021-04-06:
Well I'm compiling on Windows so not sure if I can help with that, but try to use the `t85_default` config and modify the bootloader config in `configuration/t85_default/bootloaderconfig.h`. That's what I did. Also the latest repository version doesn't even have a `t85_jumper` config.

ArminJo, 2021-04-06:
If changes to the configuration lead to an increase in bootloader size, ie you see errors like `address 0x2026 of main.bin section '.text' is not within region 'text'`, it may be necessary to change/decrease the bootloader start address as described [above](#computing-the-values) or in the *Makefile.inc*.

delay can be omitted to save memory, if external capacitance at the jumper pin is low

closed 2021-04-06 by ArminJo