ArminJo/micronucleus-firmware #24: [BUG] Device search time-out after upgrading Attiny85 with recommended settings

PinkDev1, 2022-03-09:
**Describe the bug**
Firstly, I ran a couple of test programs on an Attiny85, and they worked without problems. But after updating the bootloader using the Arduino GUI with this project, the Attiny85 doesn't seem to get recognized by Windows anymore, although the on-board LED turns on.

These are the settings I used to flash the upgrade:
[image]

_Should I have selected "16.5 MHz" (Instead of 16 MHz) before making the bootloader upgrade?_
How can I fix this?

**To Reproduce**
Steps to reproduce the behavior:
1. Set the bootloader upgrade settings according to the image
2. Tools > Burn Bootloader
3. Wait until the "Please plug in the device" message pops up
4. Plug in Attiny85 via USB
5. Wait until the process is finished
6. Unplug Attiny85
7. Wait 5 seconds
8. Plug in Attiny85
9. Wait 5 seconds  //8.5 - realize that the previously flashed program is now gone
10. Unplug the board
11. Click the upload button on the Arduino IDE
12. Wait until the "Please plug in the device" message pops up
13. Plug in the board

**Expected behavior**
That the program gets flashed into the Attiny85. 

**Used Hardware**
A regular Attiny85. connected to my laptop via USB.

**Attempted Solutions**
- Restarting Windows (Windows gives a generic "USB device not recognized" error)
- Flashing a known-to-work program (board is never detected by micronucleus)
- Flash the bootloader again (board is never detected by micronucleus)

**Desktop (please complete the following information):**
 - OS:
```
Edition	Windows 10 Home Single Language
Version	21H1
Installed on	‎2020-‎09-‎18
OS build	19043.1526
Experience	Windows Feature Experience Pack 120.2212.4170.0
```
 - Development environment: Arduino IDE v1.8.19
 - Version of the bootloader: 
   - (Latest) micronucleus-firmware v2.6.0
   - (Latest) DigistumpArduino v1.7.5

PinkDev1, 2022-03-10:
Update: Hindsight is 20/20, as usual. I re-flashed my Attiny85 using an Arduino Uno, avrdude, and ISP flashing.
In case someone makes the same mistake I did: This ZIP contains everything needed to reflash the board with the stock bootloader:
https://mega.nz/file/AhBUWZ6D#QzaX6eYttaiDf_pZmM4ep_-c48iKPguAWCM3mBXHruc

ArminJo, 2023-11-10:
There may be problems with the timing of some USB controller. Try to use an USB hub or another port or another PC.

closed 2023-11-10 by ArminJo