ArminJo/micronucleus-firmware #27: Entry on reset - cannot reprogram the board

pielgrzym, 2022-05-22:
Hi,

I'm using Attiny85 and I accidentally burned 'Entry on reset' bootloader on it. Now my program runs, but device never goes into programming mode. I connect it via usb and my program uses HID. How can I fix this? I've seen some info about D- pin, but I can't find it on any pinout diagrams :)

username7291, 2023-01-29:
With PB3/D- not connected to the pullup the device never leave programming Mode with upgrade-t85_default.hex 2.6 (1.11 ok). 
Maybe your Firmware have the same bug, but without signaling LowSpeed USB by the pullup i assume programming is too impossible.

ArminJo, 2023-11-10:
It will go into programming mode if you activate the reset pin https://github.com/ArminJo/micronucleus-firmware#attiny85-on-digispark

closed 2023-11-10 by ArminJo