ArminJo/micronucleus-firmware #30: A Ti85 with 1.11 where I can't flash 2.6 or charge program

Barabba11, 2022-11-05:
Hi, I've marked this Digispark as faulty, I don't remember right, but I feel it couldn't be recognized by usb (it powered up but didn't dialogue); after about 6 months I've connected it back, it is recognized with version 1.11:

```
 Please plug in the device ...
> Device is found!
connecting: 20% complete
connecting: 27% complete
connecting: 34% complete
connecting: 40% complete
> Device has firmware version 1.11
> Available space for user applications: 6330 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 99  page size: 64
> Erase function sleep duration: 792ms
>> Micronucleus done. Thank you!
```
and if I try to upload 2.6 (which was sucessful on other Digisparks)
```
> Please plug in the device ...
> Device is found!
connecting: 20% complete
connecting: 27% complete
connecting: 34% complete
connecting: 40% complete
> Device has firmware version 1.11
> Available space for user applications: 6330 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 99  page size: 64
> Erase function sleep duration: 792ms
parsing: 60% complete
> Erasing the memory ...
erasing: 65% complete
erasing: 70% complete
erasing: 75% complete
erasing: 80% complete
> Starting to upload ...
writing: 85% complete
writing: 90% complete
writing: 95% complete
writing: 100% complete
>> Micronucleus done. Thank you!
```

but I still have the 1.11 version, why? The Ti85 doesn't look bricked, it communicates, there are no errors on flashing. Also I can't load nothing into it, Arduino IDE returns me:
`>> Run error -1 has occured ...`
There is nothing I can do to "revive it"? Thanks

Barabba11, 2022-11-05:
Choosing another board on IDE (was Digistrump AVR boards, now ATTiny Core) I have this:
```
> Please plug in the device (will time out in 60 seconds) ... 
> Device is found!
connecting: 16% complete
connecting: 22% complete
connecting: 28% complete
connecting: 33% complete
> Device has firmware version 1.11
> Available space for user applications: 6330 bytes
> Suggested sleep time between sending pages: 8ms
> Whole page count: 99  page size: 64
> Erase function sleep duration: 792ms
parsing: 50% complete
> Erasing the memory ...
erasing: 55% complete
erasing: 60% complete
erasing: 65% complete
> Starting to upload ...
writing: 70% complete
writing: 75% complete
writing: 80% complete
> Starting the user app ...
>> Run error: Input/output error has occured ...
>> Please unplug the device and restart the program. 
La porta seriale selezionata >> Please unplug the device and restart the program. 
 non esiste o la scheda non è connessa
```

ArminJo, 2023-11-10:
The bootloader version changes AFTER the upload!

closed 2023-11-10 by ArminJo